GenPi64/gentoo-on-rpi-64bit #1: Plans Going Forward

perkinslr, 2020-11-03:
There are several considerations moving forward.

## Community organization and leadership. 
I've created a discord channel ( https://discord.gg/Eu5D2e5KJQ ) to discuss any issues that arise. 
I do not want this to become my personal project, but there is a bit of urgency in getting something visible up before people wander off and give up. At a minimum, I want two other, known community members who can take an active role in administration of the project. This should help with any personnel issues, and helps if I fail the "autobus test" at some point in the future.

## binpkg building
I have a 16 thread Ryzen machine, and probably can borrow time on two more, already set up with a cross build environment for the Pi (I have several packages customized relative to the stock version). I'm not sure what to do to establish trust, since reproducible builds are not quite there yet in gentoo.

## binpkg hosting
I have a scratch VM that I can devote to running the package mirror, it's got a 1TB/month upload limit. I don't know how popular this project is, but if it exceeds that, we'll have to figure out what to do.

ormorph, 2020-12-08:
Mesa-20.3 has been released with `vulkan` driver for rpi4. Will you have to support two build options, for rpi4 and rpi3?
If there is no real support for binhost, then at least it is necessary to update the `lite` version of the distributive.
I think that using gentoo on rpi is only relevant on rpi4. The reason is the size of the RAM.

perkinslr, 2020-12-08:
> Mesa-20.3 has been released with `vulkan` driver for rpi4. Will you have to support two build options, for rpi4 and rpi3?
> If there is no real support for binhost, then at least it is necessary to [...]

One neat thing about how Linux does it's drivers (firmware and software) is having extra ones around generally doesn't break anything.  So building mesa with the `vulkan` driver for the pi4 should work fine on the pi3.  If that proves to not be the case, we'll probably ship the `vulkan`less driver in the image, and have the `vulkan` driver via the binhost.  The packages are building with the multi-pkg feature set, so we can have as many permutations as we need to support the common targets.

So far, GenPi has only targetd 64bit pis, which would be the 3 series and 4 series.  I actually want to create a 32bit mini version, suitable for use on Pi zeroes.  (I also want to get a pi4 compute module and make sure the PCI-e stuff works properly).  The pi zero version really will be limited to what the binhost can provide, since the limited ram pretty well precludes compiling anything of note on the 0 (zram swap works, but the single core 700mhz cpu is painful).  Still, you can always stick the SD card in a desktop and use qemu for compiling packages for it.

ormorph, 2020-12-08:
Made several ebuilds for rpi, it may come in handy:
[games-misc/code-the-classics-games](https://github.com/ormorph/sleepy-tree/tree/master/games-misc/code-the-classics-games)
[app-editors/mu](https://github.com/ormorph/sleepy-tree/tree/master/app-editors/mu)
[app-editors/code-oss](https://github.com/ormorph/sleepy-tree/tree/master/app-editors/code-oss)
[media-sound/sonic-pi](https://github.com/ormorph/sleepy-tree/tree/master/media-sound/sonic-pi)
I am collecting under rpi4 8 GB.

samip5, 2020-12-12:
> Made several ebuilds for rpi, it may come in handy:
> [games-misc/code-the-classics-games](https://github.com/ormorph/sleepy-tree/tree/master/games-misc/code-the-classics-games) [...]

Could you upstream those ebuilds?

ormorph, 2020-12-12:
>Could you upstream those ebuilds?

Do you want it to be a separate archive?
These ebuilds work, I made them for myself, but it would not hurt to tweak them a bit.

perkinslr, 2020-12-12:
> > Could you upstream those ebuilds?
> 
> Do you want it to be a separate archive?
> These ebuilds work, I made them for myself, but it would not hurt to tweak them a bit.

First choice would be to get them included in `::gentoo`, both so others can benefit from it, and to keep them updated easier.  If you can't get them upstreamed yourself, we can add them to the genpi64 overlay, and we'll see about getting them upstream as time allows.

ormorph, 2020-12-12:
> If you can't get them upstreamed yourself, we can add them to the genpi64 overlay, and we'll see about getting them upstream as time allows.

I tried to add programs from raspberrypi os, so I think it's better to include them in the genpi64 overlay. 
**app-editors/code-oss** - free assembly (Visual Studio Code) for `arm`, it makes no sense to add it to gentoo.
Nobody added `PyQtChart` to gentoo.
**mu-editor** is normally installed using `pip` for `x86_64`, but this does not happen under `arm64`, which is why I made a separate ebuild.
Currently, I don't have much time, so I haven't made additions to the [trinity](https://mirror.git.trinitydesktop.org/gitea/TDE/tde-packaging-gentoo) overlay for a long time. But most likely somewhere in two months, it can be used normally.

samip5, 2020-12-12:
> I tried to add programs from raspberrypi os, so I think it's better to include them in the genpi64 overlay.
> **app-editors/code-oss** - free assembly (Visual Studio Code) for `arm`, it makes no sense [...]

Even so, it would benefit the ARM community as a whole, don't just think of the Raspberry Pi, as there are other ARM systems out there too.

perkinslr, 2020-12-12:
> I tried to add programs from raspberrypi os, so I think it's better to include them in the genpi64 overlay.
The question isn't where the programs came from, it's if they are useful on arm systems *other* than raspberry pis.

> **app-editors/code-oss** - free assembly (Visual Studio Code) for `arm`, it makes no sense to add it to gentoo.
I *think* this would make sense to use on a chromebook or other arm system.  It's probably the best candidate for upstreaming (from a usefulness perspective).
> **mu-editor** is normally installed using `pip` for `x86_64`, but this does not happen under `arm64`, which is why I made a separate ebuild.
> Currently, I don't have much time, so I haven't made [...]

And that's fine, I'm happy to include the ebuilds here for now.  If you get around to it in a few months, that'll be fine.

samip5, 2020-12-12:
Code upstreamed: https://bugs.gentoo.org/759634

ormorph, 2020-12-12:
>Even so, it would benefit the ARM community as a whole, don't just think of the Raspberry Pi, as there are other ARM systems out there too.

I have no doubt that this will benefit the community, but it takes a long time, and I don't have much time. Anyway, if you want to do this, then I do not mind.

>Code upstreamed: https://bugs.gentoo.org/759634 (although not the oss version)

This ebuild has a lot of unnecessary inheritance, so only `unpacker` would be enough. As I said, to be included in the main branch, the ebuilds must be brought to the proper form. Also I'm not sure if it will be added to the main branch.

samip5, 2020-12-12:
It might not be upstreamed as is, but that's a something still.