GenPi64/gentoo-on-rpi-64bit #2: Default Kernel?

perkinslr, 2020-12-12:
There appear to be 2 different potential sources for a default kernel, and multiple versions of each.

1. Raspberry Pi binary kernel from upstream.  Provided by `sys-kernel/raspberrypi-image`.
2. Genkernel (or some other build system) built kernel from upstream, possibly with patches applied.  Provided by `sys-kernel/raspberrypi-sources`.

Option (1) is *probably* simpler to do, but doesn't let us run gentoo-secific (or genpi64) patches.  It also should work with device-tree overlays, which may be friendlier to people coming from other Pi distros.

Option (2) takes a bit more set up, and may be more fragile, but lets us run patched kernels.  Specifically, having support for backing devices in zram would be nice, and there may be other changes.  

Anyone have any suggestions for configs, or kernel versions, or...?

samip5, 2020-12-12:
I think it would benefit most people to include the Gentoo specific patches to the upstream kernel, so probably option 2 or or last two, if we could get the binary added to the main Gentoo tree. We could also work on seeing if the Gentoo ARM team has a automated building yet, and if not, what would be needed to set one up.

Could potentially use Github actions for it.

perkinslr, 2020-12-13:
I agree that getting gentoo's patch set applied would be good.  Also having the device tree overlay I think is a must.  Personally, I want the zram module built with block device backing enabled.  

I just got done wading through the build scripts used for (3) and (4), and I don't really like them.  They predate the `sys-kernel/gentoo-kernel` and `sys-kernel/gentoo-kernel-bin` packages.  I *think* I can fork those two packages and apply the raspberry pi patchset to them, which would have an upstream-based kernel build process which allows the user to customize the `.config`, but isn't `genkernel` based (and *hopefully* will be happy without an initrd).

ormorph, 2020-12-13:
Boldly. I think it's not for nothing that Gentoo does not use special patches for `sys-kernel/raspberrypi-sources`. Genkernel is not needed to build a kernel for the raspberry pi. The usual script is enough.
I do not use binary kernel images as I need the source to build third-party modules.

perkinslr, 2020-12-13:
> Genkernel is not needed to build a kernel for the raspberry pi. The usual script is enough.

This is certainly true enough.  The issue is if you need to use the overlay system for hot patching kernel stuff to make your peripherals work.  I don't believe the `sys-kernel/raspberrypi-sources` version works with that.  That said, if you are using the pi as a low cost computer, rather than for esoteric peripherals, the stock gentoo version is likely to be superior.  No real reason we can't provide both.

ormorph, 2020-12-13:
There are third-party drivers that are not included in the kernel, but they can be built as modules. If you have a kernel source and config file, you only need to have a kernel source. It is enough to collect the kernel header files.  
```
# cd /usr/src/linux
# cp /boot/config-p4 .config
# make oldconfig
# make modules_prepare
```
For example, the built-in Wi-Fi module in the Raspberry Pi has a small network range, even with an external antenna. Using a third party wifi adapter solves this problem.

perkinslr, 2020-12-13:
I am aware that you can build kernel modules out of tree.  They go under `/lib/modules/`. I'm talking the contents of the `/boot/overlays` directory, things like `dwc-otg.dtbo`, which enables gadget mode for the pi0.

ormorph, 2020-12-13:
Not sure if gpu will work fine on the standard Gentoo kernel.  It seems that it also needs firmware.

samip5, 2020-12-13:
> Not sure if gpu will work fine on the standard Gentoo kernel. It seems that it also needs firmware.

Which GPU? The Pi4 one or Pi3 one?

ormorph, 2020-12-13:
>Which GPU? The Pi4 one or Pi3 one?

I have only Pi4..

samip5, 2020-12-13:
People are running GenPI on RPi3 and RPi4, so both should be supported.

ormorph, 2020-12-13:
>People are running GenPI on RPi3 and RPi4, so both should be supported.

Not sure if it will work as expected vc4-fkms-v3d on gentoo-sources. Reason: missing `arch/arm/boot /dts/overlays`. The file **vc4-fkms-v3d.dtbo**(vc4-fkms-v3d-overlay.dts) is missing.
But on `sys-kernel/raspberrypi-sources`, everything should work.