GenPi64/gentoo-on-rpi-64bit #3: Default Packages?

perkinslr, 2020-12-12:
Previously, which packages were installed in the image was controlled by `dev-embedded/rpi-64bit-meta`, with packages included or excluded based on use flags.  I'm not really happy with this method, but it's not the end of the world to keep using it for now (at least *some* of the packages it includes should go into `/etc/portage/sets/genpi` or similar).

But that brings up the question: What packages do we need?  I think we need 4 configurations.  Minimal, and with a DE (xfce?), as well as OpenRC and SystemD.  (Cross each of those options and you get 4 configs).  

Any packages we should include, or packages which were previously included which should be stripped?

samip5, 2020-12-12:
At least the packages supporting Wifi, needs to be shipped by default.

Like wpa_supplicant, and the Wifi firmware(?) at least.

I think we should definitely have both OpenRC and SystemD based images, as some people may want to use either of them.
I also thing that we should follow the normal Gentoo Handbook as to what to include. Like you could say there should be cron, rsyslogd and dhcp client by default

ormorph, 2020-12-18:
Of course it can and well support multiple init systems (systemd or openrc), but you need to look at the practical possibilities. First you need to create a command so that someone else is responsible for building systemd. Otherwise, the load on you will be very large if you support two assemblies at the same time. Otherwise, the project may last even less than @sakaki.
Some acquaintances using raspberry pi switched to Gentoo because of the openrc init system. They noted that the system is more stable on openrc. First you need to have at least two assemblies minimal and xfce. I liked @sakaki minimal build. For example - networkmanager, to configure wifi via `nmtui`.

ormorph, 2020-12-29:
`sys-apps/pyconfig_gen` does not use `xfce-base/xfce4-meta`, this needs to be removed from dependencies. To be able to run via `sudo`, you must first execute these:
```
$ xhost local:root
```
This can be used with other window managers...

samip5, 2020-12-29:
ntp-client needs to be included.

samip5, 2020-12-30:
Test comment

ormorph, 2021-01-10:
ffmpeg-4.3.1 no longer requires patches. You can use ffmpeg from the Gentoo upstream.  The patched version 4.2.3 performs significantly worse than the unpatched version 4.3.1. The **h264_v4l2m2m** and **h264_mmal** codecs work fine.
Eaxample:
```
ffmpeg -i input.mp4 -acodec aac -vcodec h264_v4l2m2m -b:v 2M -pix_fmt yuv420p output.mp4
```
```
ffplay -vcodec h264_mmal -i test.mp4
```
It is advisable to migrate to ffmpeg-4.3.1 from the Gentoo upstream.

jonesmz, 2021-01-10:
@ormorph 

Please create a pull request here to remove the patched ffmpeg packag: https://github.com/GenPi64/genpi64-overlay

ormorph, 2021-01-11:
> @ormorph
> 
> Please create a pull request here to remove the patched ffmpeg packag: https://github.com/GenPi64/genpi64-overlay

Ok, PR has been created. But this PR will be accepted, I don't know.

samip5, 2021-06-10:
The packages are currently being merged via profile packages which are in the genpi64-overlay at https://github.com/GenPi64/genpi64-overlay/blob/master/profiles/targets/genpi64/packages.

ormorph, 2021-06-10:
I don't like that xfce4 is added by default in the `desktop` profile. This is not good if you need to use a different desktop. Better to use a separate `desktop/xfce4` profile.

samip5, 2021-06-10:
> I don't like that xfce4 is added by default in the `desktop` profile. This is not good if you need to use a different desktop. Better to use a separate `desktop/xfce4` profile.

xfce was the default in the original image as well.

ormorph, 2021-06-10:
> xfce was the default in the original image as well.

I understand it. But if anyone wants to build a system for himself, he will choose the `lite` image. This is Gentoo, that's all its convenience.
>you can unselect it locally in /etc/portage/profile/packages, with the syntax -<package-atom>.

To do this, you need to create a new list of packages, this complicates things. I understand that this is done once, but if it is possible not to do this at all, then it would not be bad.

samip5, 2021-06-10:
> >you can unselect it locally in /etc/portage/profile/packages, with the syntax -<package-atom>.
> 
> To do this, you need to create a new list of packages, this complicates things. I understand that this is done once, but if it is possible not to do this at all, then it would not be bad.
> 

Not entirely accurate. You need to create it and include ONLY the packages you DONT want selected.

ormorph, 2021-06-10:
>Not entirely accurate. You need to create it and include ONLY the packages you DONT want selected.

Yes, I checked it, it works. Thanks.

jonesmz, 2021-06-10:
If you feel strongly about this still. I don't see why we wouldn't welcome the creation of profiles for alternative desktops beyond xfce.
I can't say that we'll build official images for them, but it seems reasonable to have the alternative profiles.

If you make a PR I would review it for you.

ormorph, 2021-06-10:
>I can't say that we'll build official images for them, but it seems reasonable to have the alternative profiles.

Basically, I only needed a basic desktop profile to install the [TDE](https://github.com/ormorph/TDE) `desktop`.
It makes sense to use alternative profiles for such Plasma or Gnome, and for lightweight desktops, the basic `desktop` profile is enough.

ormorph, 2021-06-18:
I noticed that on the latest images, the root user password is not created at the first start. The reason is in the security policy specified in `/etc/security/passwdqc.conf`. As a result, the raspberrypi64 password is not available for use by the root user. This needs to be fixed.
See: `man passwdqc.conf`.

samip5, 2021-06-18:
> I noticed that on the latest images, the root user password is not created at the first start. The reason is in the security policy specified in `/etc/security/passwdqc.conf`. As a result, the [...]

No, that's incorrect assumption. The root password has been swaped to `GenPi@@` which is done here: https://github.com/GenPi64/Build.Dist/blob/alpha8.5/subtargets/gentoo-base.json#L40 and the parser is here: https://github.com/GenPi64/Build.Dist/blob/alpha8.5/parsers/password/password

ormorph, 2021-06-18:
>The root password has been swaped to GenPi@

Hmm, I understood. But in principle, nothing prevents you from doing this on systemd.
```
[Unit]
Description=Сhanges password 
After=syslog.target

[Service]
Type=oneshot
ExecStart=/usr/lib/systemd/scripts/chpass start
RemainAfterExit=yes
StandardOutput=syslog

[Install]
WantedBy=multi-user.target
```
You can just write a `chpass` script in bash.

samip5, 2021-06-18:
> Hmm, I understood. But in principle, nothing prevents you from doing this on systemd.
>
> ```
> [Unit]
> Description=Сhanges password
> After=syslog.target
>
> [Service]
> Type=oneshot [...]

Yes, but rather than maintaining unit files for openrc and systemd, it gets set via the password parser by our build script.

ormorph, 2021-06-18:
>Yes, but rather than maintaining unit files for openrc and systemd, it gets set via the password parser by our build script.

Yes, this is all correct, but then it is necessary to correct the description(password) and remove the password change from the /etc/init.d/autoexpand-root script. I do not know how in the systemd assembly the resizing of partitions was carried out on the first start. I am using openrc.

ormorph, 2021-07-01:
Have you tested btrfs on microsd?
I checked it on two microsd, each has problems after updating the ports:
```
# emaint sync -a 
```
after that, this command shows errors:
```
# btrfs check --readonly /dev/sdb2
```
[btrfs.log.gz](https://github.com/GenPi64/gentoo-on-rpi-64bit/files/6746079/btrfs.log.gz)

When using the ext4 filesystem, these problems do not arise.

samip5, 2021-07-02:
> Have you tested btrfs on microsd?
> I checked it on two microsd, each has problems after updating the ports:

Could you eloborate, after doing updating what?

ormorph, 2021-07-02:
>Could you eloborate, after updating what?

No, this happens on a freshly installed Genpi64, after performing write operations on it. I gave this command as an example:
```
# emaint sync -a 
```
I just noticed that the microsd, I quickly got out of order. I started to figure out what was the matter and came to the conclusion that btrfs is not very compatible with microsd. 
For some reason, the FS table becomes unusable.

samip5, 2021-07-04:
> No, this happens on a freshly installed Genpi64, after performing write operations on it. I gave this command as an example:
>
> ```
> # emaint sync -a
> ```
>
> I just noticed that the microsd, I quickly [...]

Hmm, are you sure that it's not your SD card itself? Most of us are using USB SSDs and haven't noticed issues with it.

ormorph, 2021-07-04:
>Hmm, are you sure that it's not your SD card itself?

This does not happen on ext4. The command shows that everything is fine:
```
# e2fsck -f /dev/sdb2
```
I copied the files to the ext4 partition and corrected the initramfs, there were no problems with the flash drive. 
If I have a 16 gigabyte microsd old, then the second 64 gigabyte microsd is completely new and outputs the same. MicroSD for 16 gigabytes, when installing a new gcc on btrfs, it generally went into read-only mode. If the cause of the errors is in microsd, then on ext4 this should have happened again. Well, if you immediately check after installation, then the check shows that the btrfs partition is normal, and after writing to it, errors appear.
Checked on two different rpi4.

ormorph, 2021-07-05:
Hmm, it looks like the problem is in the FS mount parameter: `subvol=/`. I repartitioned the btrfs partition, created a new subvolume, and then added the FS mount option: `subvol=@rootfs`. After completing these steps, btrfs began to work stably without errors. Checked both in chroot mode and in rpi4. It turns out btrfs is unstable when using `subvol=/`.

samip5, 2021-07-05:
> Hmm, it looks like the problem is in the FS mount parameter: `subvol=/`. I repartitioned the btrfs partition, created a new subvolume, and then added the FS mount option: `subvol=@rootfs`. After [...]

Did you need to do anything fancy to the btrfs partition or is `@rootfs` valid by default?

ormorph, 2021-07-05:
>Did you need to do anything fancy to the btrfs partition or is @rootfs valid by default?

/etc/fstab :
```
PARTUUID="5ad7aa00-02"  /       btrfs   noatime,compress=zstd,ssd,discard,subvol=@rootfs      00
```
/boot/cmdline.txt:
```
console=serial0,115200 console=tty1 dwc_otg.lpm_enable=0 root=PARTUUID=5ad7aa00-02 rootfstype=btrfs fsck.repair=no usbhid.mousepoll=0 rootwait=10 init=/sbin/init
```
/usr/src/initramfs-1.0.2/bin/init:
```
rootflags = 'compress=zstd:3,noatime,subvol=@rootfs'
```
Something like this...
Typically microsd was corrupted when using chroot mode, but sometimes without it.

jonesmz, 2021-07-06:
> Hmm, it looks like the problem is in the FS mount parameter: `subvol=/`. I repartitioned the btrfs partition, created a new subvolume, and then added the FS mount option: `subvol=@rootfs`. After [...]

That shouldn't have any effect on whether the filesystem has errors.

The btrfs log that you attached implies that your sd card was written incorrectly.

I have yet to encounter a btrfs filesystem error on any of the images I've written to my SD cards.

ormorph, 2021-07-06:
>The btrfs log that you attached implies that your sd card was written incorrectly.

After writing the image to microsd, there are no errors. Typically, errors appear after trying to build in the chroot environment. 
rpimount.sh:
```
#!/bin/bash

DEVICE=$1

if [ -z ${DEVICE} ] ; then
        echo no such device
        exit 0
fi

if ! [ -d "/mnt/root" ] ; then
        mkdir /mnt/root
fi

echo "${DEVICE}"2

mount -t btrfs -o noatime,compress=zstd,ssd,discard,subvol=/ "${DEVICE}"2 /mnt/root/
mount "${DEVICE}"1 /mnt/root/boot

mount --rbind /dev /mnt/root/dev
mount --rbind /sys /mnt/root/sys
mount -t proc none /mnt/root/proc
mount --rbind /binpkg /mnt/root/var/cache/binpkgs
mount --rbind /var/cache/distfiles /mnt/root/var/cache/distfiles
mount --bind /program/tmp/portage /mnt/root/var/tmp/portage
```
Run command:
```
# ./rpimount.sh /dev/sdb
# cp /etc/resolv.conf /mnt/root/etc/
# chroot /mnt/root
```
Have you tried updating this way? 
Sometimes errors appear even after not very long work on rpi4. It looks like specifying `/` for the `subvol` parameter was a bad idea. Using a separate subvolume disables access to the upper areas of the btrfs partition.

jonesmz, 2021-07-06:
> Have you tried updating this way?

Yes, i haven't had a problem.

> Using a separate subvolume disables access to the upper areas of the btrfs partition.

Disables access? In what way? If you mean at the logical block level, then no, that's not how btrfs works. If you mean at the vfs level, then yes it does, but only until you mount a different subvolume at a different location.

I have no objection to replacing / with rootfs, if that's a built-in BTRFS concept. I just don't see how this has anything to do with the btrfs subvolume mount option.

jonesmz, 2021-07-06:
Re-reading the comment history, it appears that "@rootfs" is not a built in btrfs thing. It's a new subvolume that you've created specifically for your example.

ormorph, 2021-07-06:
>Re-reading the comment history, it appears that "@rootfs" is not a built in btrfs thing. It's a new subvolume that you've created specifically for your example.

Quite right, this is a separate subvolume, from under which there is no access to the upper volume. That is why it is safer.

jonesmz, 2021-07-06:
> Quite right, this is a separate subvolume, from under which there is no access to the upper volume. That is why it is safer.

What do you mean by safer?

ormorph, 2021-07-06:
There are several advantages, for example, you can use snapshots. As far as I understand, a separate table is used for the subvolume, which does not affect the top one table. Well, I have confirmed this in practice, I no longer have errors.
This is most likely due to some types of microsd.
I recorded the image quite simply:
```
# dd if=genpi-aarch64-lite-latest.img of=/dev/sdb status=progress
```

jonesmz, 2021-07-18:
Hrmm, I apologise, I thought it had responded to this, but must have had bad cell reception or something at the time.

I'm definitely not opposed to making the root filesystem a non "/" subvolume. That has a variety of benefits that I think are worth having since the cost is basically free at runtime.

However, I strongly suspect that you are misdiagnosing this problem. Unless there is a bug in btrfs when it comes to micro sd cards, the subvolumes should have nothing to do with the errors you had.

If you think that the subvolumes are directly influencing these bugs, i strongly recommend you bring them up on the linux kernel developers mailing list for BTRFS after trying with the latest git release of the BTRFS tools.

ormorph, 2021-07-18:
>Unless there is a bug in btrfs when it comes to micro sd cards, the subvolumes should have nothing to do with the errors you had.

Hmm, subvolumes do not affect this error, but this error occurs when subvolumes are not used. `subvol=/` is the use of the root partition of the disk without subvolumes. As I previously found similar examples of errors on the network, there the problem was associated with compression and the use of btrfs without subvolumes. 
I would only be glad if I was wrong about these mistakes. However, these errors are repeated fairly consistently. On the newly labeled microsd, the command shows that the microcd is normal:
```
# badblocks -v /dev/sdb

Checking blocks 0 to 15441919
Checking for bad blocks (read-only test):
done
Pass completed, 0 bad blocks found. (0/0/0 errors)
```
As far as I understand, these errors appear only on some devices, for example, I do not have such problems on ssd, but these problems are present with microcd.

jonesmz, 2021-07-18:
It's possible that something is wrong with the image generated by our build servers, potentially.

Could you see what happens if you make a new subvolume in a freshly installed genpi64 image, and mount that?

There's certainly going to be more to this bug than meets the eye.

ormorph, 2021-07-18:
Later I will check, I will also download the latest Genpi64 image again, maybe that changes.

ormorph, 2021-07-18:
Checked, after creating a new subvolume, no errors occur. But also after downloading the ports, errors appear again:
```
# emaint sync -a
```
The image is not corrupted, sha512 matches.
```
# sha512sum genpi-aarch64-lite-latest.img.zst
e58a96c3dde12e64046dab832c8ae8375bb78dc734f960c4cb7a0e57cf122b84f4284c8c007eb735879e3a3a9596c14015dfcc31c12c6b284de03ecbe457d39d  genpi-aarch64-lite-latest.img.zst
```

jonesmz, 2021-07-20:
I meant that it's possible that the image itself, before you write it to the SD card, was generated incorrectly.

I think that's unlikely, but it's possible.

ormorph, 2021-07-21:
>I meant that it's possible that the image itself, before you write it to the SD card, was generated incorrectly.

Yes, it looks like it is. The actions that I did:
```
# zstd -d genpi-aarch64-lite-latest.img.zst
# dd if=genpi-aarch64-lite-latest.img of=/dev/sdb status=progress
# mkfs.btrfs -f /dev/sdb2
# losetup -Pf genpi-aarch64-lite-latest.img
# mount -t btrfs -o noatime,compress=zstd,ssd,clear_cache,discard,subvol=/ /dev/loop0p2 /mnt/gentoo/
# mount -t btrfs -o noatime,compress=zstd,ssd,clear_cache,discard,subvol=/ /dev/sdb2 /mnt/root/
# cp -ax /mnt/gentoo/. /mnt/root/
# umount /mnt/gentoo/
# umount /mnt/root
```
Next, check, insert it into rpi4 and run. Everything booted the size of the partition increased.
Next, I again insert microsd into it as a USB stick and run the mount script attached above:
```
# ./rpimount.sh /dev/sdb
# cp /etc/resolv.conf /mnt/root/etc/
# chroot /mnt/root
```
Next, I run the check:
```
# btrfs check --readonly /dev/sdb2
Opening filesystem to check...
Checking filesystem on /dev/sdb2
UUID: 3e318e51-8bcc-4f77-9543-e9f0bcc28117
[1/7] checking root items
[2/7] checking extents
[3/7] checking free space cache
[4/7] checking fs roots
[5/7] checking only csums items (without verifying data)
[6/7] checking root refs
[7/7] checking quota groups skipped (not enabled on this FS)
found 1230454784 bytes used, no error found
total csum bytes: 951968
total tree bytes: 255115264
total fs tree bytes: 248283136
total extent tree bytes: 5357568
btree space waste bytes: 34621341
file data blocks allocated: 975339520
 referenced 2071535616
```
The check was successful, there are no errors. There was also a problem with the root password, since it was not created. The reason I gave above is to allow short passwords in the `/etc/security/passwdqc.conf` file. Well, you can also make the /etc/init.d/autoexpand-root script more versatile. So that you can use both btrfs and ext4 without problems. 
[autoexpand-root.gz](https://github.com/GenPi64/gentoo-on-rpi-64bit/files/6852898/autoexpand-root.gz)

ormorph, 2021-07-21:
I think that the use of subvolumes is still necessary so that snapshots can be used.

samip5, 2021-07-21:
> I think that the use of subvolumes is still necessary so that snapshots can be used.

Snapshots work fine in the current form, I believe.

ormorph, 2021-07-21:
>Snapshots work fine in the current form, I believe.

I don't even want to check here, because I find it inconvenient to store snapshots of the root partition in the root partition.

jonesmz, 2021-07-21:
If you make a pull request that causes the image to be built with the root partition 's default snapshot to be different than /, i would review it for you, and help you get it merged.

ormorph, 2022-01-23:
To build chromium, you can use the patches from [here](https://bugs.gentoo.org/786597).
You can add this to `/etc/portage/patches/...`
Checked against clang-13, chromium-96.

ormorph, 2022-05-19:
I noticed that the latest build has CFLAGS parameters in /etc/portage/make.conf. There is no `-ftree-vectorize` flag, it seems to work slower without it. Why is this compiler flag not enabled for the build now?

jonesmz, 2022-05-19:
This seems like something that should be in its own issue number.

Can you open a new issue here on github to discuss?

ormorph, 2022-05-19:
Ok. Opened  #32