GenPi64/gentoo-on-rpi-64bit #8: Image for  Raspberry 400?

pwasiewi, 2021-01-30:
Has anyone tested Raspberry 400 with gentoo?
Could you put here some instructions (links?) for making image for it?

perkinslr, 2021-01-30:
We only can support hardware we have access to.  If someone *has* one of these and is willing to test and report issues with it, we're happy to resolve any issues it has with GenPi (and the base system should "just work").

ormorph, 2021-02-18:
>Has anyone tested Raspberry 400 with gentoo?

The Raspberry 400 is based on Pi4, so there shouldn't be any problems. The Pi4 may experience wifi issues. For a solution, you will need to install the files from [here](https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm/). You need the `brcmfmac43455-sdio.bin` file and copy it to the /lib/firmware/brcm/ directory. Failed to get this to work on `sys-kernel/linux-firmware-20201218` had to be masked.

I like the Pi4 option better, because with [this](https://www.aliexpress.com/item/32963307586.html) cooling, it is quite possible to build gentoo on a Pi4 without using a third-party PC. The 8GiB version builds even `dev-qt/qtwebengine` quite well.

perkinslr, 2021-02-18:
> For a solution, you will need to install the files from [here](https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm/). You need the `brcmfmac43455-sdio.bin` file and copy it to the [...]

In theory, the wifi ucode is available from `sys-firmware/raspberrypi-wifi-ucode::gentoo`.  It's also possible to use `savedconfig` with `linux-firmware` to install other linux firmware files, but not the 43455 broadcom stuff.  That said, on a Pi4, nothing in `linux-firmware` appears to actually be used.  That would change on a compute module with pcie-expansion stuff, but even there it's probably 50/50 on actually needing `linux-firmware`.

ormorph, 2021-02-18:
I installed `firmware/raspberrypi-wifi-ucode`, but I had to abandon it in favor of brcmfmac43455-sdio.bin.  Wi-Fi worked, but at the time of the load on the processor it was malfunctioning.  For example, the access point disappeared from time to time. This was happening at compile time. It may be necessary to create a 9999 version of this package.

G10h4ck, 2022-01-23:
AFAIU GenPI64 provided images boot fine on Raspberry Pi 400, while a bit of trickery is needed to make the WiFi work. Am I right?

ormorph, 2022-01-23:
No, it was a problem when migrating from old builds from sakaki.
There are problems with btrfs in current builds. It is recommended that after writing to the USB flash drive, re-format the btrfs partition and copy the data from the image to the formatted partition.
Most likely the problem is related to the fact that the image was formed using qemu.

G10h4ck, 2022-01-23:
So I have downloaded the image from here
https://fi.packages.genpi64.com/genpi-aarch64-desktop-latest.img.zst
The main link give 404 so I have used the mirror

if after flashing de SD i execute

`btrfstune -u /dev/MYPARTITION`

It doesn't boot anymore, basically the kernel panic because it is not able to find the root partition, so I re-flashed without altering the UUID, this way it boot without problem, but doesn't login automatically as per README, inserting demouser and the password it login without fuss.

Then in the terminal I see that both wlan' and eth' are listed, wlan0 is up (haven't tested if it really connect), eth0 is down, if I try to bring it up with `ip link set dev eth0 up` it fail and in `dmesg` some message about not being capable of finding the devie appears toghether with other bcmXXXXX messages that seems related, maybe the ethernet chip need a different driver then raspberrypi 4B ?

G10h4ck, 2022-01-23:
I have moved the same SD to another Raspberry Pi 400 I have and there eth0 is up at boot without dmesg errors, the only difference that I know the two Pi400 have is that one is the one with spanish keyboard and the other have italian keyboard, so
PI400-es eth0 works fine with both raspberrypiOS and ubuntu, but not with genpi-aarch64-desktop-latest
PI400-it eth0 works fine with both raspberrypiOS and ubuntu, seems to work genpi-aarch64-desktop-latest (I'll test it more and then edit the comment)

ormorph, 2022-01-23:
You also need to check the btrfs section after writing to it, for example, after updating the ports tree.
See the problem [here.](https://github.com/GenPi64/gentoo-on-rpi-64bit/issues/3#issuecomment-883887048)
>As I expected on this kind of device lspci -vv doesn't output anything interesting about the ethernet card any suggestion on how further investigate what can it be the difference between the two device and contribute back?

```
$ cat /sys/class/net/eth0/device/uevent
DRIVER=bcmgenet
OF_NAME=ethernet
OF_FULLNAME=/scb/ethernet@7d580000
OF_COMPATIBLE_0=brcm,bcm2711-genet-v5
OF_COMPATIBLE_N=1
OF_ALIAS_0=ethernet0
```

```
$ zgrep BCMGENET /proc/config.gz
CONFIG_BCMGENET=y
```