GenPi64/gentoo-on-rpi-64bit #10: GenPI 64 lite image doesn't boot

enx1998, 2021-02-14:
Hi,
I have tried to install the lite image alpha 5, but it doesn't boot. 
I have decompressed the zst image and them used dd to write it to sd card. After that I have put the sd card in the raspberry pi 4 8GB and when I plug in the power cord I can see the pi bootloader and then a blank screen. I have never seen the "rainbow square".
With the sakaki's image the pi boot fine even with an usb hard disk.

Thanks ENx

perkinslr, 2021-02-14:
We've been struggling a bit with getting the boot partition happy without intervention.  Make sure the `lbo` flag is set on partition 1.  Make sure the kernel image exists (the -p4 or -pi4 variant), and make sure the config.txt file has a `kernel=kernel8-p4.img` (or -pi4 if that's the name of the kernel image, don't remember what it was under alpha5).  Also make sure the cmdline.txt is correct, it should say root=PARTUUID=xxxxxxxx-02

If bandwidth isn't a major issue, alpha6 is a bit easier to get off the ground than alpha5, and alpha7 is in progress.  It *should* just work out of the box (as in I don't plan to release it until it just works).

enx1998, 2021-02-14:
hi perkinslr, thanks for your reply, I realize I'm using alpha6, although the Readme show the latest lite image as alpha 5. I make it work: in config.txt I don't have kernel config. Now it boot fine.
Thanks for the great work ENx

perkinslr, 2021-02-14:
Glad you got it working.  Need to figure out how to automate updating the readme as versions change.  Hopefully version changes will slow down slightly as we get more stable...

perkinslr, 2021-03-01:
Alright, I'm going to close this as we've moved on to alpha7, which has a 0-steps bootable image.  Please re-open or start a new issue if there are further failure-to-boot issues.

closed 2021-03-01 by perkinslr