GenPi64/gentoo-on-rpi-64bit #15: Emerge and profiles not working

Forkmaster14, 2021-05-19:
When I try to emerge, it says /etc/portage/make.conf is not a valid symlink and it doesn't work. Online it says I should change my profile, but when I try to do that, it says failed to get a list of valid profiles, quitting. Any solutions?

pingyofdoom, 2021-05-31:
I had the same thing, I had to run etc-update

samip5, 2021-06-10:
Which image is this?

chraaroe, 2021-06-28:
Hi @samip5, perhaps I can contribute to this. I believe I'm experiencing the same issue as @Forkmaster14.

I've just gotten a new Raspberry Pi 4b to replace my 3b which had been running Sakaki's Gentoo for quite some time.

Following your guide and installing the latest Lite image, I find myself with a configuration which is complaining a bit.  Doing an initial `etc-update` fails with the following message:

```
!!! Section 'gentoo' in repos.conf has location attribute set to nonexistent directory: '/var/db/repos/gentoo'
!!! Section 'genpi-tools' in repos.conf has location attribute set to nonexistent directory: '/var/db/repos/sakaki-tools'
!!! Section 'genpi64' in repos.conf has location attribute set to nonexistent directory: '/var/db/repos/genpi64'
!!! Invalid Repository Location (not a dir): '/var/db/repos/gentoo'

!!! /etc/portage/make.profile is not a symlink and will probably prevent most merges.
!!! It should point into a profile within /var/db/repos/gentoo/profiles/
!!! (You can safely ignore this message when syncing. It's harmless.)

!!! Your current profile is invalid. If you have just changed your profile
!!! configuration, you should revert back to the previous configuration.
!!! Allowed actions are limited to --help, --info, --search, --sync, and
!!! --version.
```

Not sure about how to progress right now. Still looking into it.

samip5, 2021-06-28:
> Hi @samip5, perhaps I can contribute to this. I believe I'm experiencing the same issue as @Forkmaster14.
>
> I've just gotten a new Raspberry Pi 4b to replace my 3b which had been running Sakaki's [...]

You need to run `emerge --sync` before trying to do anything.

chraaroe, 2021-06-28:
Thanks for the quick reply, @samip5.

I've been away from Gentoo for too long... How could I forget. A bit embarrassed by that, actually.

closed 2021-06-29 by samip5