GenPi64/gentoo-on-rpi-64bit #16: GCC isn't compiling

pingyofdoom, 2021-05-25:
I just updated to alpha8, genup isn't completing past GCC on a Raspberry Pi 4. I've tried 3 times now, I haven't timed it but I think it's a 4-5 hour compile.  I can't find the swap file, is it still getting installed after the update to BTRFS?

samip5, 2021-06-10:
There is no swapfile by default, one should use zram as there's built-in support for it in kernel.

https://wiki.gentoo.org/wiki/Zram

samip5, 2021-06-18:
> I just updated to alpha8, genup isn't completing past GCC on a Raspberry Pi 4. I've tried 3 times now, I haven't timed it but I think it's a 4-5 hour compile. I can't find the swap file, is it still getting installed after the update to BTRFS?

Actually, came across this too. You need to change MAKEOPTS, it's trying to do 16 jobs more than likely.