GenPi64/gentoo-on-rpi-64bit #22: Links to images in the doc are broken

gagara, 2022-01-21:
Simply
```
$ wget -c https://packages.genpi64.com/genpi-aarch64-lite-latest.img.zst
--2022-01-21 09:19:39--  https://packages.genpi64.com/genpi-aarch64-lite-latest.img.zst
SSL_INIT
Resolving packages.genpi64.com... 45.55.3.11, 2604:a880:1:20::20:d001
Connecting to packages.genpi64.com|45.55.3.11|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://s3.genpi64.com/binpkgs/genpi-aarch64-lite-latest.img.zst [following]
--2022-01-21 09:19:41--  https://s3.genpi64.com/binpkgs/genpi-aarch64-lite-latest.img.zst
SSL_INIT
Resolving s3.genpi64.com... 91.232.155.81, 2001:67c:1be8:1337::443
Connecting to s3.genpi64.com|91.232.155.81|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-01-21 09:19:41 ERROR 404: Not Found.
```

jlpoolen, 2022-01-23:
I tried making a link, it broke.  Then I copy pasted the wget commands from the documentation.  They broke:

    gentooVM1 /home/jlpoole/GenPi64 #  wget -c https://packages.genpi64.com/genpi64-lite-latest.img.zst
    --2022-01-22 10:18:46--  https://packages.genpi64.com/genpi64-lite-latest.img.zst
    Resolving packages.genpi64.com... 45.55.3.11, 2604:a880:1:20::20:d001
    Connecting to packages.genpi64.com|45.55.3.11|:443... connected.
    HTTP request sent, awaiting response... 302 Found
    Location: https://s3.genpi64.com/binpkgs/genpi64-lite-latest.img.zst [following]
    --2022-01-22 10:18:47--  https://s3.genpi64.com/binpkgs/genpi64-lite-latest.img.zst
    Resolving s3.genpi64.com... 91.232.155.81, 2001:67c:1be8:1337::443
    Connecting to s3.genpi64.com|91.232.155.81|:443... connected.
    HTTP request sent, awaiting response... 404 Not Found
    2022-01-22 10:18:49 ERROR 404: Not Found.

    gentooVM1 /home/jlpoole/GenPi64 # wget -c https://packages.genpi64.com/genpi-aarch64-lite-latest.img.zst
    --2022-01-22 10:19:36--  https://packages.genpi64.com/genpi-aarch64-lite-latest.img.zst
    Resolving packages.genpi64.com... 45.55.3.11, 2604:a880:1:20::20:d001
    Connecting to packages.genpi64.com|45.55.3.11|:443... connected.
    HTTP request sent, awaiting response... 302 Found
    Location: https://s3.genpi64.com/binpkgs/genpi-aarch64-lite-latest.img.zst [following]
    --2022-01-22 10:19:36--  https://s3.genpi64.com/binpkgs/genpi-aarch64-lite-latest.img.zst
    Resolving s3.genpi64.com... 91.232.155.81, 2001:67c:1be8:1337::443
    Connecting to s3.genpi64.com|91.232.155.81|:443... connected.
    HTTP request sent, awaiting response... 404 Not Found
    2022-01-22 10:19:37 ERROR 404: Not Found.

    gentooVM1 /home/jlpoole/GenPi64 #  wget -c https://packages.genpi64.com/genpi-aarch64-desktop-latest.img.zst
    --2022-01-22 10:21:09--  https://packages.genpi64.com/genpi-aarch64-desktop-latest.img.zst
    Resolving packages.genpi64.com... 45.55.3.11, 2604:a880:1:20::20:d001
    Connecting to packages.genpi64.com|45.55.3.11|:443... connected.
    HTTP request sent, awaiting response... 302 Found
    Location: https://s3.genpi64.com/binpkgs/genpi-aarch64-desktop-latest.img.zst [following]
    --2022-01-22 10:21:09--  https://s3.genpi64.com/binpkgs/genpi-aarch64-desktop-latest.img.zst
    Resolving s3.genpi64.com... 91.232.155.81, 2001:67c:1be8:1337::443
    Connecting to s3.genpi64.com|91.232.155.81|:443... connected.
    HTTP request sent, awaiting response... 404 Not Found
    2022-01-22 10:21:09 ERROR 404: Not Found.

    gentooVM1 /home/jlpoole/GenPi64 #

Visiting genpi64.com causes the user to be redirected to: https://s3.genpi64.com/binpkgs/ which then displays XML.  A sample <contents> element reveals that there is life on this server as a package was built as recently as January 18, 2022.  This is hopeful news that this project has not been abandoned.  So it looks like there has been some changes in the directory structure of the server under genpi64.com.

    <Contents>
        <Key>Packages</Key>
        <LastModified>2022-01-18T06:59:37.269Z</LastModified>
        <ETag>"e311ce5c589828157d9eac2838db6a01"</ETag>
        <Size>563169</Size>
        <Owner>
            <ID>
            </ID>
            <DisplayName>minio</DisplayName>
        </Owner>
        <StorageClass>STANDARD</StorageClass>
    </Contents>
    <Contents>

A search of the XML listing using "aarch64" produces nothing, so the images are not in this tree or at least they are not being disclosed.  I'll try the RTC Chat to see if someone can help; in the meantime I'll log to this Issue as it merits attention.

jlpoolen, 2022-01-23:
Ahah!   Clicking the "mirror" link in the table of downloads results in a prompt for allowing a download.  Looks like the problem concerns a primary server and the work-around is to try a "mirror" server, e.g. https://fi.packages.genpi64.com/genpi-aarch64-lite-latest.img.zst

jlpoolen, 2022-01-23:
This problem is also affecting, I think, the build.sh for the alpha branch; see #25

samip5, 2022-01-23:
This is known, as we are in the progress of migrating to an s3 object storage, and the image files haven't been uploaded there as we are in the process of getting newer images built and then uploaded.

jlpoolen, 2022-01-23:
What is the estimated date/time for the migration/rebuild to be completed?

samip5, 2022-01-23:
The current "latest" image is really old, like 6 months+ old so that's why it wasn't uploaded. In the next few weeks I think.

PR #27 (2022-01-24): Refactor README
Also added the new alpha9 image link.

Fixes #22
closed 2022-01-24 by samip5