GenPi64/gentoo-on-rpi-64bit #23: Invitation To Participate in Discord Fails

jlpoolen, 2022-01-23:
This project states on 1/22/2022:

      Open an issue here, or join the Discord server at https://discord.gg/a9VqTGHPy6

I click through to Discord and I accept the invite:

[image]

When I am in the forum, I select the lobby hash.  I'm told I cannot send messages on the channel (#lobby).  There ought to be instructions on what to do in order to participate, or something needs to be changed on the Discord side.
[image]

Otherwise, as it is now, there's no point to clicking through to Discord.

jonesmz, 2022-01-23:
Sorry about the trouble.

Here's a new link : https://discord.gg/BHbMbUBHjZ

jlpoolen, 2022-01-23:
I tried the link: https://discord.gg/BHbMbUBHjZ

Had the same experience, when I arrived, and even after selecting "#lobby", I have the same dark letters stating I may not post.

jonesmz, 2022-01-23:
Oh. My mistake.

You have to click the green checkbox. Doing that will let you access the other chat rooms

Its a spam prevention tool

perkinslr, 2022-01-23:
@jlpoolen If you need further assistance feel free to send me a DM on discord; I @mentioned you in the channel you need to use.

jlpoolen, 2022-01-23:
I direct messaged perkinslr.  For propriety, here is the screen I see and the only green box, Like/Disklike, that I checked.
[image]

jlpoolen, 2022-01-23:
I played around with clicking the check box for Carl-bot and noticed the column to the left changing and I that was in #bugs, so I tried ?#general" and it looks like I'm in the discussion/channel/discord/group.
Thank you.

closed 2022-01-23 by jlpoolen