GenPi64/gentoo-on-rpi-64bit #24: genpi-aarch64-lite-latest (Jan 22, 2022) No Start on Rasp. Pi Zero 2 W

jlpoolen, 2022-01-23:
I have a Raspberry Pi Zero 2 W, released October 2021, and tried to run the genpi-aarch64-lite-latest.

The colorized initial screen appears and then nothing more, no command-line like console or output as appears during a successful boot.

The Raspberry Pi Zero 2 uses the Cortex-A53 processor inside a Broadcom  BCM2710A1.  The Raspberry Pi 3 model B and  Raspberry Pi 3 model B+ both use the same Cortex-A53 processor; however, the model 3 chips are at a higher frequency.  Nonetheless, the Rapsberry Pi 3 and the Raspberry Pi Zero 2 W both use the same instruction set: ARMv8-A (64/32-bit) 

So all indications suggest that this binary blob should work on the Raspberry Pi Zero 2 W. 

I have a dedicated virtual machine for this project where I'm running Gentoo 64 and I'm installing a cross compiler.  Could my problem be that binary blobs for the loader are different for the Pi Zero 2 vs. Pi 3 Model B and the point of failure is in the loading, not necessarily the kernel?

jlpoolen, 2022-01-23:
The genpi-aarch64-lite-latest was downloaded today, Jan 22, 2022, from the Finland mirror.

    gentooVM1 /home/jlpoole/GenPi64 # wget -c https://fi.packages.genpi64.com/genpi-aarch64-lite-latest.img.zst
    --2022-01-22 11:08:47--  https://fi.packages.genpi64.com/genpi-aarch64-lite-latest.img.zst
    Resolving fi.packages.genpi64.com... 91.232.155.81, 2001:67c:1be8:1337::443
    Connecting to fi.packages.genpi64.com|91.232.155.81|:443... connected.
    HTTP request sent, awaiting response... 200 OK
    Length: 565494847 (539M)
    Saving to: ‘genpi-aarch64-lite-latest.img.zst’

    genpi-aarch64-lite-latest. 100%[=======================================>] 539.30M  4.84MB/s    in 86s

    2022-01-22 11:10:13 (6.29 MB/s) - ‘genpi-aarch64-lite-latest.img.zst’ saved [565494847/565494847]

    gentooVM1 /home/jlpoole/GenPi64 #   zstd -d genpi-aarch64-lite-latest.img.zst -o genpi-aarch64-lite-latest.img
    genpi-aarch64-lite-latest.img.zst: 3342860288 bytes
    gentooVM1 /home/jlpoole/GenPi64 # ls -la
    total 1515648
    drwxr-xr-x 2 root    root          4096 Jan 22 11:13 .
    drwxr-xr-x 4 jlpoole jlpoole       4096 Jan 22 10:17 ..
    -rw-r--r-- 1 root    root    3342860288 Jan 22 11:13 genpi-aarch64-lite-latest.img
    -rw-r--r-- 1 root    root     565494847 Apr 27  2021 genpi-aarch64-lite-latest.img.zst
    gentooVM1 /home/jlpoole/GenPi64 #

I copied the image onto a 64GB SDHC:

    gentooVM1 /home/jlpoole/GenPi64 #  dd if=genpi-aarch64-lite-latest.img of=/dev/xvda4 bs=1M status=progress; sync
    3337617408 bytes (3.3 GB, 3.1 GiB) copied, 265 s, 12.6 MB/s
    3188+0 records in
    3188+0 records out
    3342860288 bytes (3.3 GB, 3.1 GiB) copied, 302.957 s, 11.0 MB/s
    gentooVM1 /home/jlpoole/GenPi64 #

jlpoolen, 2022-01-23:
Checked the **graphics specifications** to see if possible difference:  **there is a match**.

https://en.wikipedia.org/wiki/Raspberry_Pi#Specifications

     GPU row:
         RPi 3 Model B: Broadcom VideoCore IV @ 250 MHz[b] or for the "+": Broadcom VideoCore IV @ 400 MHz (Core) / 300 MHz (V3D)
         Zero 2 W: Broadcom VideoCore IV @ 400 MHz (Core) / 300 MHz (V3D)

The Pi Zero 2 W matches the RPi Model 3B+.

jonesmz, 2022-01-23:
I have been trying to get this working. I can't figure out what the problem is.

jlpoolen, 2022-01-23:
Do you have any build notes posted somewhere?  It's been years since I have cross compiled, I did for the Plug, Genesi Efika & Beaglebone prior to 2015.  I've just emerged cross-dev and I am scratching my head on how to specify the ARMv8-A 64 bit.  I'm not finding "crossdev -t help" completely helpful.  (I was put-out by Vapier's high handedness and impatience at times, brilliant as he was.)  I decided I would try the ARMv8 in 64 bit mode realizing there have been problems with cameras working in 64bit Bullseye --- I'm hoping the problems magically solve themselves in a Gentoo environment.

jlpoolen, 2022-01-23:
I'm suspecting the problem may be with the bootloaders and/or proprietary blobs.  But I like challenges.

jonesmz, 2022-01-23:
Under this org on github. Look at the "Build.Dist" repository.

Use the alpha9 branch of that repo. And run "build.sh" ro build the image.

It should build everything without intervention

jlpoolen, 2022-01-23:
I'm a Subversion fan, but I thought I'd be brave and attempt to use git.  I think I succeeded, but perhaps the problem I encountered here is not related to my properly selecting the alpha9 branch?

    gentooVM1 /home/jlpoole # git clone https://github.com/GenPi64/Build.Dist.git
    Cloning into 'Build.Dist'...
    remote: Enumerating objects: 1888, done.
    remote: Counting objects: 100% (795/795), done.
    remote: Compressing objects: 100% (371/371), done.
    remote: Total 1888 (delta 441), reused 553 (delta 316), pack-reused 1093
    Receiving objects: 100% (1888/1888), 274.05 KiB | 1.08 MiB/s, done.
    Resolving deltas: 100% (1073/1073), done.
    gentooVM1 /home/jlpoole # cd Build.Dist/
    gentooVM1 /home/jlpoole/Build.Dist # ls
    GenPi64OpenRC.json         GentooAMD64.json  build.sh   env.sh        requirements.txt
    GenPi64OpenRCDesktop.json  LICENSE.txt       chroot.sh  parsers       scripts
    GenPi64Systemd.json        README.md         config     patchtargets  subtargets
    gentooVM1 /home/jlpoole/Build.Dist # git switch
    fatal: missing branch or commit argument
    gentooVM1 /home/jlpoole/Build.Dist # git switch alpha9
    Branch 'alpha9' set up to track remote branch 'alpha9' from 'origin'.
    Switched to a new branch 'alpha9'
    gentooVM1 /home/jlpoole/Build.Dist # ls
    GenPi64Generic.json        GenPi64SystemdDesktop.json     README.md  env.sh            scripts
    GenPi64OpenRC.json         GentooAMD64.json               build.sh   parsers           subtargets
    GenPi64OpenRCDesktop.json  GentooAMD64OpenRCDesktop.json  chroot.sh  patchtargets
    GenPi64Systemd.json        LICENSE.txt                    config     requirements.txt
    gentooVM1 /home/jlpoole/Build.Dist # ./build.sh
    Traceback (most recent call last):
      File "/home/jlpoole/Build.Dist/parsers/includejson/includejson", line 7, in <module>
        import lockfile
    ModuleNotFoundError: No module named 'lockfile'
    run complete.
    gentooVM1 /home/jlpoole/Build.Dist # whoami
    root
    gentooVM1 /home/jlpoole/Build.Dist # date
    Sat Jan 22 13:28:37 PST 2022
    gentooVM1 /home/jlpoole/Build.Dist #

jlpoolen, 2022-01-23:
Okay, got past that hurdle, I had to:

      emerge dev-python/lockfile

jlpoolen, 2022-01-23:
Next hurdle: " **xz: (stdin): File format not recognized**"

    gentooVM1 /home/jlpoole/Build.Dist # ./build.sh
    loading manifest
    load complete
    Loading status
    status not loadable
    load complete
    [{'name': 'gentoo-base', 'deps': [''], 'parser': 'includejson', 'args': ['subtargets/gentoo-base.json', 'patchtargets/gentoo-aarch64.json', 'patchtargets/gentoo-openrc.json']}, {'name': 'genpi64-lite-finalize', 'deps': ['gentoo-base'], 'parser': 'includejson', 'args': ['subtargets/genpi64-lite-finalize.json']}, {'name': 'echo-lite-banner', 'deps': ['genpi64-lite-finalize'], 'parser': 'printf', 'args': ['############################\n# Build complete\n# To install your image on an SD card, run\n# dd if=/path/to/image of=/path/to/sdcard\n#############################\n']}]
    checking  gentoo-base
    checking  genpi64-lite-finalize
    checking  echo-lite-banner
    completed= ['']
    manifest= [{'name': 'gentoo-base', 'deps': [], 'parser': 'includejson', 'args': ['subtargets/gentoo-base.json', 'patchtargets/gentoo-aarch64.json', 'patchtargets/gentoo-openrc.json']}, {'name': 'genpi64-lite-finalize', 'deps': ['gentoo-base'], 'parser': 'includejson', 'args': ['subtargets/genpi64-lite-finalize.json']}, {'name': 'echo-lite-banner', 'deps': ['genpi64-lite-finalize'], 'parser': 'printf', 'args': ['############################\n# Build complete\n# To install your image on an SD card, run\n# dd if=/path/to/image of=/path/to/sdcard\n#############################\n']}]
    checking  gentoo-base deps []
    all deps satisfied for gentoo-base
    executing  gentoo-base deps []
    checking  echo-lite-banner deps ['genpi64-lite-finalize']
    running jobs [(<Popen: returncode: None args: ['/home/jlpoole/Build.Dist/parsers/includejso...>, 'gentoo-base')]
    Using sub-completed file
    loading manifest
    loading patch patchtargets/gentoo-aarch64.json
    loading patch patchtargets/gentoo-openrc.json
    load complete
    Loading status
    status not loadable
    load complete
    [{'name': 'downloadstage3', 'deps': [''], 'parser': 'downloadstage3', 'args': ['']}, {'name': 'stage3', 'deps': ['downloadstage3'], 'parser': 'stage3', 'args': ['']}, {'name': 'qemu', 'before': 'config', 'deps': ['', 'stage3'], 'parser': 'qemu', 'args': ['']}, {'name': 'config', 'deps': ['stage3', 'qemu'], 'parser': 'config', 'args': ['']}, {'name': 'overlays', 'deps': ['config'], 'parser': 'overlays', 'args': ['']}, {'name': 'git', 'deps': ['overlays'], 'parser': 'git', 'args': ['']}, {'name': 'portage', 'deps': ['git'], 'parser': 'portage', 'args': ['']}, {'name': 'etc', 'deps': ['portage'], 'parser': 'etc', 'args': ['']}, {'name': 'removeoldgcc', 'deps': ['etc'], 'parser': 'echo', 'args': ['skipping --unmerge sys-devel/gcc:10.2.0 -1']}, {'name': 'enablebinpkg', 'deps': ['removeoldgcc'], 'parser': 'enablefeature', 'args': ['buildpkg', 'binpkg-multi-instance']}, {'name': 'rebuildinstalled', 'deps': ['enablebinpkg'], 'parser': 'emerge', 'args': ['@installed --jobs --emptytree', '-elogind -filecaps']}, {'name': 'zapcertifi', 'deps': ['rebuildinstalled'], 'parser': 'emerge', 'args': ['--unmerge dev-python/certifi']}, {'name': 'certifi', 'deps': ['zapcertifi'], 'parser': 'emerge', 'args': ['dev-python/certifi', 'python_targets_python3_8']}, {'name': 'screen', 'deps': ['certifi'], 'parser': 'emerge', 'args': ['-v1g app-misc/screen', '', '']}, {'name': 'update_portage', 'deps': ['screen'], 'parser': 'emerge', 'args': ['-v1bgj portage']}, {'name': 'locale', 'deps': ['update_portage'], 'parser': 'rawcommand', 'args': ['eselect locale set en_US.utf8']}, {'name': 'updateudev', 'deps': ['', 'locale'], 'before': 'users', 'parser': 'emerge', 'args': ['--update eudev']}, {'name': 'users', 'deps': ['config', 'update_portage', 'updateudev'], 'parser': 'users', 'args': ['']}, {'name': 'eix', 'deps': ['update_portage'], 'parser': 'emerge', 'args': ['eix']}, {'name': 'eixupdate', 'deps': ['eix'], 'parser': 'rawcommand', 'args': ['eix-update']}, {'name': 'bison', 'deps': ['eixupdate'], 'parser': 'emerge', 'args': ['-v1 bison']}, {'name': 'updatelibc', 'deps': ['bison'], 'parser': 'emerge', 'args': ['-v1bg --update --deep  --newuse --jobs glibc']}, {'name': 'emergesystem', 'deps': ['updatelibc'], 'parser': 'emerge', 'args': ['-v1bg --update --deep --newuse --jobs $(/em-config jobs) --load-average $(/em-config load-average) --tree --autounmask-backtrack=y --backtrack=3000 @system', '-X sqlite -introspection -gtk -elogind -policykit -qt5 -qt4']}, {'name': 'setprofile', 'deps': ['emergesystem', 'config'], 'parser': 'setprofile', 'args': ['']}, {'name': 'emergeprofile', 'deps': ['setprofile'], 'parser': 'emerge', 'args': ['-v1bg --update --deep --newuse --jobs $(/em-config jobs) --load-average $(/em-config load-average) --tree --autounmask-backtrack=y --backtrack=3000 @system']}, {'name': 'addsets', 'deps': ['emergeprofile'], 'parser': 'addsets', 'args': ['']}, {'name': 'setuptools', 'deps': ['addsets'], 'parser': 'emerge', 'args': ['-v1bg --update --newuse dev-python/setuptools']}, {'name': 'emergeworld', 'deps': ['setuptools'], 'parser': 'emerge', 'args': ['-v1bg --deep --update --newuse --with-bdeps=y --keep-going --jobs $(/em-config jobs) --load-average $(/em-config load-average) --tree --autounmask-backtrack=y --backtrack=3000 @world'], 'env': {'SUPPRESS_BASHRC_MAKEOPTS': '1'}}, {'name': 'depclean', 'deps': ['emergeworld'], 'parser': 'emerge', 'args': ['-v --depclean']}, {'name': 'setupservices', 'deps': ['depclean'], 'parser': 'setupse [... cut ...]
    [I] app-arch/xz-utils
         Available versions:  5.2.5-r1^t **9999*l^t {+extra-filters nls split-usr static-libs ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32"}
         Installed versions:  5.2.5-r1(00:53:41 01/13/22)(extra-filters nls split-usr -static-libs ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="64 -32 -x32")
         Homepage:            https://tukaani.org/xz/
         Description:         utils for managing LZMA compressed files

    gentooVM1 /home/jlpoole/Build.Dist #

jonesmz, 2022-01-23:
Sorry I'm not currently at home, so its a bit difficult to help much.

I am.... Surprised that it doesn't like your xz version. This might be easier to help you with on discord, since we have several users who could assist.

I asked one of the other devs to figure out why you can't post on discord.

samip5, 2022-01-23:
> Next hurdle: " **xz: (stdin): File format not recognized**"
>
>     gentooVM1 /home/jlpoole/Build.Dist # ./build.sh
>     loading manifest
>     load complete
>     Loading status
>     status not loadable [...]

This seems to be a reault of incomplete stage3 archive.

jlpoolen, 2022-01-23:
I've determined the problem: it is a result of the server migration and/or redirect:

	gentooVM1 /home/jlpoole/temp # tar -xapf stage3-arm64.tar.xz 
	xz: (stdin): File format not recognized
	tar: Child returned status 1
	tar: Error is not recoverable: exiting now
	gentooVM1 /home/jlpoole/temp # ls -la stage3-arm64.tar.xz 
	-rw-r--r-- 1 root root 5278 Jan 23 00:00 stage3-arm64.tar.xz
	gentooVM1 /home/jlpoole/temp # cat stage3-arm64.tar.xz 
	<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
	<html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en-ca">
	  <head>
	    <title>Bytemark - Mirror</title>
	    <link rel="shortcut icon" href="/common/favicon.ico" />
	    <link href="/common/general.css" rel="stylesheet" type="text/css" />
	    <link href="/common/subpage.css" rel="stylesheet" type="text/css" />
	    <link href="/common/navigation.css" rel="stylesheet" type="text/css" />
	   </head>
	  <body id="hosting_overview">

	
	<div id="wrapper">

	
	

	<div id="mainarea">

	

	  <div id="toparea">
	      [image]
	      <span id="mirrorstext">Public Mirrors</span>
	  </div>

	

	<div id="contentarea">
	<div id="mirror">

	  <table>
	   <tr><th valign="top">[image]</th><th>Name</th><th>Last modified</th><th>Size</th></tr>
	   <tr><th colspan="4"><hr /></th></tr>
	<tr><td valign="top">[image]</td><td><a href="/gentoo/releases/arm64/">Parent Directory</a></td><td>&nbsp;</td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="20211219T233639Z/">20211219T233639Z/</a></td><td align="right">2021-12-20 03:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="20211226T235150Z/">20211226T235150Z/</a></td><td align="right">2021-12-27 03:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="20220102T233648Z/">20220102T233648Z/</a></td><td align="right">2022-01-03 04:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="20220109T232200Z/">20220109T232200Z/</a></td><td align="right">2022-01-10 04:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="20220116T233655Z/">20220116T233655Z/</a></td><td align="right">2022-01-17 03:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="README">README</a></td><td align="right">2015-10-18 02:22  </td><td align="right">372 </td></tr>
	<tr><td valign="top">[image]</td><td><a href="current-install-arm64-minimal/">current-install-arm64-minimal/</a></td><td align="right">2022-01-17 03:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="current-stage3-arm64-musl-hardened/">current-stage3-arm64-musl-hardened/</a></td><td align="right">2022-01-17 03:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="current-stage3-arm64-musl/">current-stage3-arm64-musl/</a></td><td align="right">2022-01-17 03:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="current-stage3-arm64-systemd/">current-stage3-arm64-systemd/</a></td><td align="right">2022-01-17 03:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="current-stage3-arm64/">current-stage3-arm64/</a></td><td align="right">2022-01-17 03:01  </td><td align="right">  - </td></tr>
	<tr><td valign="top">[image]</td><td><a href="latest-install-arm64-minimal.txt">latest-install-arm64-minimal.txt</a></td><td align="right">2022-01-23 04:00  </td><td align="right">133 </td></tr>
	<tr><td valign="top">[image]</td><td><a href="latest-iso.txt">latest-iso.txt</a></td><td align="right">2022-01-23 04:00  </td><td align="right">133 </td></tr>
	<tr><td valign="top">[image]</td><td><a href="latest-stage3-arm64-musl-hardened.txt">latest-stage3-arm64-musl-hardened.txt</a></td><td align="right">2022-01-23 04:00  </td><td align="right">141 </td></tr>
	<tr><td valign="top">[image]</td><td><a href="latest-stage3-arm64-musl.txt">latest-stage3-arm64-musl.txt</a></td><td align="right">2022-01-23 04:00  </td><td align="right">132 </td></tr>
	<tr><td valign="top">[image]</td><td><a href="latest-stage3-arm64-systemd.txt">latest-stage3-arm64-systemd.txt</a></td><td align="right">2022-01-23 04:00  </td><td align="right">135 </td></tr>
	<tr><td valign="top">[image]</td><td><a href="latest-stage3-arm64.txt">latest-stage3-arm64.txt</a></td><td align="right">2022-01-23 04:00  </td><td align="right">127 </td></tr>
	<tr><td valign="top">[image]</td><td><a href="latest-stage3.txt">latest-stage3.txt</a></td><td align="right">2022-01-23 04:00  </td><td align="right">346 </td></tr>
	   <tr><th colspan="4"><hr /></th></tr>
	</table>
	</div>
	</div>
	</div>

	</div>
	</body>
	</html>
	gentooVM1 /home/jlpoole/temp #

samip5, 2022-01-23:
I'm currently building a new image on an arm64 Ampere instance, should take ~6-24h to be completed.

jlpoolen, 2022-01-23:
Thank you.  Could you update this issue when the task has completed and then I'll try again?

samip5, 2022-01-23:
> Thank you. Could you update this issue when the task has completed and then I'll try again?

Sure, will do.

samip5, 2022-01-24:
@jlpoolen I have not tested this, but you can download it from: https://s3.genpi64.com/binpkgs/genpi64-arm64-openrc-lite-alpha9.img.zst

jlpoolen, 2022-03-24:
I obtained an image from Mike Jones and tested it on a Raspberry Pi Zero 2 W.  It worked!

I therefore had to build my own image using this project which I forked.  I succeeded, thanks to Mike's and Logan Perkin's help.   I used a fresh Gentoo VM on a Xen hypervisor (sharing Gentoo kernel) to build the image.  The hypervisor is an AMD Bulldozer 8120, I dedicated 16 CPUs to the VM and 16GB of ram.  

The machine the built the image:

     jlpoole@jenk ~/Build.Dist $ uname -a
     Linux jenk 5.15.11-gentoo-x86_64 #1 SMP Sun Jan 16 09:24:16 -00 2022 x86_64 AMD FX(tm)-8120 Eight-Core Processor AuthenticAMD GNU/Linux
     jlpoole@jenk ~/Build.Dist $

The image:

     jlpoole@GenPi64 ~ $ uname -a
     Linux GenPi64 5.10.92-v8 #1 SMP PREEMPT Sat Mar 19 15:33:55 -00 2022 aarch64 GNU/Linux
     jlpoole@GenPi64 ~ $

I did not use a cross-compiler is distributed compiling with other machines. The build took approximately **8 days & nights**, I'm told that's because everything was being done in [QEMU ](https://www.qemu.org/)-- an open source emulator.  My next step is I'm building a cross compiler on the Bulldozer hypervisor, I'll also build a cross compiler on a newer AMD Ryzen hypervisor and then try this project, again, with two cross compilers and see what performance gains are made.

Here are photos of a RaspberryPiZero2W I own which may used in this project to advertise the fact that Gentoo now runs on a $15 computer.

[image]
[image]

closed 2022-03-24 by jlpoolen