GenPi64/gentoo-on-rpi-64bit #26: genpi64:default/linux/arm64/17.0/genpi64 profile really need to depend on networkmanager ?

G10h4ck, 2022-01-24:
I understand this could be needed or a sane default for the desktop profile, but is it really needed on the non-desktop one? Shouldn't we remove it?

ormorph, 2022-01-24:
This is to make it user friendly. `nmtui` only works in the console. This is useful when setting up Wi-Fi.

G10h4ck, 2022-01-24:
Thanks didn't know that trick!

jonesmz, 2022-01-24:
I think it would be best if we moved away from having the profiles depend on specific packages, as it makes it difficult for people who prefer alternatives to use them.

Better would be to use sets, or simply have them pre-installed in the world set.

samip5, 2022-01-24:
> Better would be to use sets, or simply have them pre-installed in the world set.

We were previously using sets, and it was a nightmare with those. I don't agree with that.

jonesmz, 2022-01-25:
How was a set for 'default apps' more of a nightmare than a profile, which sets a bunch of accept_keywords and package_use flags?

Profiles should have nothing to do with which applications are installed. Do the profiles in gentoo upstream dictate installed applications? Or only use flags?

samip5, 2022-01-25:
> How was a set for 'default apps' more of a nightmare than a profile, which sets a bunch of accept_keywords and package_use flags?
> 
> Profiles should have nothing to do with which applications are installed. Do the profiles in gentoo upstream dictate installed applications? Or only use flags?

It was a nightmare as the sets were not handled by Git.