GenPi64/gentoo-on-rpi-64bit #31: emerge --sync crashes on new install

garyemiller, 2022-04-20:
```
# emerge --sync
>>> Syncing repository 'gentoo' into '/var/db/repos/gentoo'...
/usr/bin/git fetch origin --depth 1
 * Using keys from /usr/share/openpgp-keys/gentoo-release.asc
 * Refreshing keys via WKD ...                                           [ ok ]
 * Trusted signature found on top commit
=== Sync completed for gentoo
>>> Syncing repository 'genpi64' into '/var/db/repos/genpi64'...
/usr/bin/git clone --depth 1 --single-branch --branch alpha9 https://github.com/GenPi64/genpi64-overlay.git .
Cloning into '.'...
warning: Could not find remote branch alpha9 to clone.
fatal: Remote branch alpha9 not found in upstream origin
!!! git clone error in /var/db/repos/genpi64
>>> Syncing repository 'genpi-tools' into '/var/db/repos/genpi-tools'...
/usr/bin/git fetch origin --depth 1
=== Sync completed for genpi-tools
!!! Repository 'genpi64' is missing masters attribute in '/var/db/repos/genpi64/metadata/layout.conf'
!!! Set 'masters = gentoo' in this file for future compatibility

 * IMPORTANT: 3 config files in '/etc' need updating.
 * See the CONFIGURATION FILES and CONFIGURATION FILES UPDATE TOOLS
 * sections of the emerge man page to learn how to update config files.
Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.9/emerge", line 51, in <module>
    retval = emerge_main()
  File "/usr/lib/python3.9/site-packages/_emerge/main.py", line 1318, in emerge_main
    return run_action(emerge_config)
  File "/usr/lib/python3.9/site-packages/_emerge/actions.py", line 3778, in run_action
    return action_sync(emerge_config)
  File "/usr/lib/python3.9/site-packages/_emerge/actions.py", line 2388, in action_sync
    success, msgs = syncer.auto_sync(options=options)
  File "/usr/lib/python3.9/site-packages/portage/emaint/modules/sync/sync.py", line 90, in auto_sync
    return self._sync(repos, return_messages, emaint_opts=options)
  File "/usr/lib/python3.9/site-packages/portage/emaint/modules/sync/sync.py", line 264, in _sync
    display_news_notification(
  File "/usr/lib/python3.9/site-packages/portage/proxy/objectproxy.py", line 31, in __call__
    return result(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/_emerge/post_emerge.py", line 42, in display_news_notification
    news_counts = count_unread_news(portdb, vardb)
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 471, in count_unread_news
    count = manager.getUnreadItems(repo, update=True)
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 214, in getUnreadItems
    self.updateItems(repoid)
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 174, in updateItems
    if item.isRelevant(
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 289, in isRelevant
    if restriction.checkRestriction(**kwargs):
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 399, in checkRestriction
    return kwargs["profile"].startswith(self.profile[:-1])
AttributeError: 'NoneType' object has no attribute 'startswith'
```

jonesmz, 2022-04-20:
```
usr/bin/git clone --depth 1 --single-branch --branch alpha9 https://github.com/GenPi64/genpi64-overlay.git .
Cloning into '.'...
```

Wrong branch. Switch to master.

garyemiller, 2022-04-20:
This is from an untouched, absolutely hot off the press download!

garyemiller, 2022-04-20:
genup also fails, multiple ways:

```
# genup
>>> Syncing repository 'gentoo' into '/var/db/repos/gentoo'...
/usr/bin/git fetch origin --depth 1
 * Using keys from /usr/share/openpgp-keys/gentoo-release.asc
 * Refreshing keys via WKD ...                                           [ ok ]
 * Trusted signature found on top commit
=== Sync completed for gentoo
>>> Syncing repository 'genpi64' into '/var/db/repos/genpi64'...
/usr/bin/git clone --depth 1 --single-branch --branch alpha9 https://github.com/GenPi64/genpi64-overlay.git .
Cloning into '.'...
warning: Could not find remote branch alpha9 to clone.
fatal: Remote branch alpha9 not found in upstream origin
!!! git clone error in /var/db/repos/genpi64
>>> Syncing repository 'genpi-tools' into '/var/db/repos/genpi-tools'...
/usr/bin/git fetch origin --depth 1
=== Sync completed for genpi-tools
!!! Repository 'genpi64' is missing masters attribute in '/var/db/repos/genpi64/metadata/layout.conf'
!!! Set 'masters = gentoo' in this file for future compatibility

 * IMPORTANT: 3 config files in '/etc' need updating.
 * See the CONFIGURATION FILES and CONFIGURATION FILES UPDATE TOOLS
 * sections of the emerge man page to learn how to update config files.
Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.9/emerge", line 51, in <module>
    retval = emerge_main()
  File "/usr/lib/python3.9/site-packages/_emerge/main.py", line 1318, in emerge_main
    return run_action(emerge_config)
  File "/usr/lib/python3.9/site-packages/_emerge/actions.py", line 3778, in run_action
    return action_sync(emerge_config)
  File "/usr/lib/python3.9/site-packages/_emerge/actions.py", line 2388, in action_sync
    success, msgs = syncer.auto_sync(options=options)
  File "/usr/lib/python3.9/site-packages/portage/emaint/modules/sync/sync.py", line 90, in auto_sync
    return self._sync(repos, return_messages, emaint_opts=options)
  File "/usr/lib/python3.9/site-packages/portage/emaint/modules/sync/sync.py", line 264, in _sync
    display_news_notification(
  File "/usr/lib/python3.9/site-packages/portage/proxy/objectproxy.py", line 31, in __call__
    return result(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/_emerge/post_emerge.py", line 42, in display_news_notification
    news_counts = count_unread_news(portdb, vardb)
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 471, in count_unread_news
    count = manager.getUnreadItems(repo, update=True)
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 214, in getUnreadItems
    self.updateItems(repoid)
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 174, in updateItems
    if item.isRelevant(
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 289, in isRelevant
    if restriction.checkRestriction(**kwargs):
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 399, in checkRestriction
    return kwargs["profile"].startswith(self.profile[:-1])
AttributeError: 'NoneType' object has no attribute 'startswith'
* Checking Portage configuration, please wait...
!!! Repository 'genpi64' is missing masters attribute in '/var/db/repos/genpi64/metadata/layout.conf'
!!! Set 'masters = gentoo' in this file for future compatibility
WARNING: One or more repositories have missing repo_name entries:

	/var/db/repos/genpi64/profiles/repo_name

NOTE: Each repo_name entry should be a plain text file containing a
unique name for the repository on the first line.

fatal: not a git repository (or any of the parent directories): .git
* Gentoo System Updater v1.0.29
* Updating Portage tree and syncing the eix cache
* (this may take some time)...
 * Syncing all portage overlays
/usr/bin/eix-sync: line 396: layman: command not found
 * layman -S failed
[... 32 lines cut ...]
    return self._sync(repos, return_messages, emaint_opts=options)
  File "/usr/lib/python3.9/site-packages/portage/emaint/modules/sync/sync.py", line 264, in _sync
    display_news_notification(
  File "/usr/lib/python3.9/site-packages/portage/proxy/objectproxy.py", line 31, in __call__
    return result(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/_emerge/post_emerge.py", line 42, in display_news_notification
    news_counts = count_unread_news(portdb, vardb)
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 471, in count_unread_news
    count = manager.getUnreadItems(repo, update=True)
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 214, in getUnreadItems
    self.updateItems(repoid)
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 174, in updateItems
    if item.isRelevant(
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 289, in isRelevant
    if restriction.checkRestriction(**kwargs):
  File "/usr/lib/python3.9/site-packages/portage/news.py", line 399, in checkRestriction
    return kwargs["profile"].startswith(self.profile[:-1])
AttributeError: 'NoneType' object has no attribute 'startswith'
 * emerge --sync failed
 * Time statistics:
    16 seconds for syncing
    16 seconds total

* genup: Error: Caught signal - exiting
```

garyemiller, 2022-04-20:
> ```
> usr/bin/git clone --depth 1 --single-branch --branch alpha9 https://github.com/GenPi64/genpi64-overlay.git .
> Cloning into '.'...
> ```
> 
> Wrong branch. Switch to master.

And where would I do that?

garyemiller, 2022-04-20:
removing "--branch alpha9" from /etc/portage/repos/genpi64.conf fixed it 

Should be fixed upstream.

jonesmz, 2022-04-20:
Yes. Well, when I have time.

samip5, 2022-04-30:
> genup works better after that, but still expects layman is installed, which it is not.

That's an upstream thing, I asked about it and got turned down "because it's expected to be used with layman" was the answer pretty much.

garyemiller, 2022-04-30:
Then maybe layman should be installed by default?

closed 2022-10-13 by samip5