GenPi64/gentoo-on-rpi-64bit #34: sync issue [fix]

metaphysicsIO, 2022-09-28:
I was having an error when trying to update. The error was basically that the branch didn't exist.

It was corrected when I changed the `sync-git-clone-extra-opts` line in /etc/portage/repos.conf/genpi64.conf

`#sync-git-clone-extra-opts = --single-branch --branch alpha9` to `sync-git-clone-extra-opts = --single-branch --branch master`.

jonesmz, 2022-09-28:
You installed an older image that was for development testing.

Newer nearly-nightly-build images are available https://s3.genpi64.com/images/

E.g. https://s3.genpi64.com/images/nullGenPi64Systemd.img.zst

Join our discord for more prompt updates: https://discord.gg/BHbMbUBHjZ

samip5, 2022-09-28:
> You installed an older image that was for development testing.
> 
> Newer nearly-nightly-build images are available https://s3.genpi64.com/images/
> 
> E.g. https://s3.genpi64.com/images/nullGenPi64Systemd.img.zst
> 
> Join our discord for more prompt updates: https://discord.gg/BHbMbUBHjZ

The issue is that the README has links to alpha9 version. :)

metaphysicsIO, 2022-09-29:
Oh. I had corrected the error myself and just updated everything myself. This is good to know, we should update the README file as per @samip5

jonesmz, 2022-09-29:
the github has all the dev activity. Just hard to miss announcement style things if we dont update the readme file.

closed 2022-10-08 by samip5