GenPi64/gentoo-on-rpi-64bit #35: Autoexpand_root not working?

jlpoolen, 2022-10-10:
Downloaded a newly built image, nullGenPi64OpenRC-08-10-22.img.zst, from https://s3.genpi64.com/images/
Burned onto a 16 GB diskette.

I saw on the first boot-up screen some output generated from the autoexpand_root.sh.

I started adding some packages and ran out of disk space.

I am unable to reconcile the fact that I should have 14GBs devoted to the Linux root, yet I only seem to have 3.1 GB according to **df**. 

I remember running into something like this issue before and @Jonesmz on the Discord channel helped.

Also, before a fix is run, is there something I should run that will exhibit the space mismatch more succinctly so that in the future other users can use such a sequence of commands to determine whether the auto-expansion has worked?

```

GenPi64B /home/jlpoole # fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 14.84 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xcd541400

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *      2048   524287   522240  255M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      524288 31116287 30592000 14.6G 83 Linux
GenPi64B /home/jlpoole # df -lh
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  3.1G  2.3G  279M  90% /
devtmpfs         10M     0   10M   0% /dev
tmpfs           1.6G 1012K  1.6G   1% /run
cgroup_root      10M     0   10M   0% /sys/fs/cgroup
shm             3.9G     0  3.9G   0% /dev/shm
/dev/mmcblk0p1  255M   72M  184M  28% /boot
tmpfs           782M     0  782M   0% /run/user/1001
GenPi64B /home/jlpoole #
```
The image does have sys-fs/btrfs-progs installed:

```
GenPi64B /home/jlpoole # eix btrfs
[I] sys-fs/btrfs-progs
     Available versions:  5.15.1(0/0)^t ~5.18.1-r1(0/0)^t ~5.19.1(0/0)^t **9999(0/0)*l^t {+convert doc +man python reiserfs static static-libs udev +zstd PYTHON_SINGLE_TARGET="python3_8 python3_9 python3_10"}
     Installed versions:  5.15.1(0/0)^t(04:01:34 AM 09/27/2022)(convert static-libs zstd -doc -python -reiserfs -static PYTHON_SINGLE_TARGET="python3_10 -python3_8 -python3_9")
     Homepage:            https://btrfs.wiki.kernel.org
     Description:         Btrfs filesystem utilities

* sys-fs/btrfsmaintenance
     Available versions:  *0.4.2 ~0.5 {systemd}
     Homepage:            https://github.com/kdave/btrfsmaintenance
     Description:         Scripts for btrfs maintenance tasks like periodic scrub, balance, trim or defrag

Found 2 matches
GenPi64B /home/jlpoole #
```

jlpoolen, 2022-10-10:
I cannot seem to login to Pastebin; nonetheless, I was able to post my dmesg output anonymously:
https://pastebin.com/CvkK94tB

jlpoolen, 2022-10-10:
Did some research, and found: https://wiki.gentoo.org/wiki/Btrfs#Resizing

This worked for me:   **btrfs filesystem resize max /**

(Nice feature about btrfs that is can expand "live"!)

```
GenPi64B /home/jlpoole # btrfs filesystem resize max /
Resize device id 1 (/dev/mmcblk0p2) from 3.09GiB to max
GenPi64B /home/jlpoole # df -lh
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2   15G  2.3G   12G  17% /
devtmpfs         10M     0   10M   0% /dev
tmpfs           1.6G 1012K  1.6G   1% /run
cgroup_root      10M     0   10M   0% /sys/fs/cgroup
shm             3.9G     0  3.9G   0% /dev/shm
/dev/mmcblk0p1  255M   72M  184M  28% /boot
tmpfs           782M     0  782M   0% /run/user/1001
GenPi64B /home/jlpoole #
```
Is this something that can be automated, or should the documentation be updated to indicate this is a once-in-a-lifetime statement that should be executed on a new deployment of an image?

samip5, 2022-10-10:
> Did some research, and found: https://wiki.gentoo.org/wiki/Btrfs#Resizing
>
> This worked for me: **btrfs filesystem resize max /**
>
> (Nice feature about btrfs that is can expand "live"!)
>
> ```
> GenPi64B [...]

The script in /boot should have done that on boot, but somehow is failing only the btrfs portion of it as the partition itself seems to have largened, just not the filesystem itself.