GenPi64/gentoo-on-rpi-64bit #38: Who's Behind The Curtain

jlpoolen, 2022-10-29:
This project does not identify the principals behind it.  While I think I know who they are having learned from interactions in their Discord channel, I think it behooves the project to identify those people to enhance its credibility.

samip5, 2022-11-23:
Could you please explain as to what is this issue really about?

perkinslr, 2022-11-24:
It's a fair question with a not-completely-trivial anwswer.  

This github organization and the genpi64 domain names are owned and controlled by LP-Programming L.L.C., my software consultancy firm.  The discord server is controlled by my personal discord account.  

The individual projects hosted within the github organization are mixed in ownership and origin, like most projects that move around on github.  Generally, absent a copyright assignment, work for hire agreement, or similar, the individual creative contributions are authored and owned by the entity writing them.  It can get murkey who owns exactly what piece of code, and what degree of collaboration is required to switch from individual pieces to joint-authorship.  It generally doesn't matter, as they are all under free licesnses.  If you want to see who is involved, check out the individual projects' commit logs.

Of note, the core of the Build.Dist project was created under joint authorship with a local (now bought out) security company.  The contributions by others post upload-to-github are in the same category as any of the other projects.  In theory, ADC could take the initial upload to github, plus any commits authored by me, and distribute / use / licsense them under any terms they like, under the US collaborative-authorship rules.  They *would* have to get permission from any contributors *other* than me to use commits authored by anyone else (and getting into the fair-use mess).  In practice, their build-system has moved on from this as part of their acquision of the local company, so it's unlikely to come up (which was part of why they were willing to donate the skeleton to this project in the first place).  

Anyway, we use github in order to keep the whole project progress externally visible, so that questions of "who is responsible" matter less.  If you don't want to run some random image posted online, build the project yourself.  That said, creating something of an "about" page, with at least some basic information about the major contributors to the project, is on my todo list.  If for no other reason than to allow the principal contributors to cross-polinate with any other projects they have.

jlpoolen, 2022-12-29:
Thank you for the explanation. It seems that the Jenkins build service and Amazon Web Services are an integral part of the project now.  Who is subsidizing and/or providing those?  Are they something that could just disappear or become unavailable?  Getting people to come on board requires a bit confidence that the ship (this project) will still be floating in the near future, and/or arrangements can be made to provide services such as the build engine.

For instance, I've succeeded in getting Google's Coral USB accelerator working on Gentoo on and an AMD64 laptop and a Raspberry Pi 4B.  So I am more focused now on this project now that I can run Coral on a Raspberry product running Gentoo -- a Linux I'm most familiar with and feel is the most adaptable for my needs.  When I took a look at the status of the Jenkins build system, I was told on Discord that Gentoo upstream broke the system and its been that way since October 2022.  So, now more than ever, I am interested in who is providing the Jenkins system and what can be done to help new on-boarders, such as myself, help.

samip5, 2022-12-29:
> It seems that the Jenkins build service and Amazon Web Services

Where do you see AWS? The s3 object storage is on MiniO, not on AWS if that's what you're referring to.
Jenkins and the s3 object storage via MiniO are both backed by Kapsi internet-users association's servers, that's been around from 2003 and not going anywhere. 

I host those both on my membership there as member's need to have a connection to Finland to be able to use those services.

More details about Kapsi internet-users association in English: https://kapsi.fi/english.html.