GenPi64/gentoo-on-rpi-64bit #39: Over The Rainbow

woofwoof, 2022-12-30:
Apologies for reaching out but any ideas please where I should start to get around a persistent 'rainbow screen' on boot on a Raspberry Pi Zero 2 W? The only alternative I tried which also failed was a PINN install. I did amend the recommended file there to account for devices (a TFT screen & fan) attached to the GPIO - could those peripherals be the root of the persistent rainbow on a native install (using Balena Etcher under Windows)? Thanks again for any help...

jonesmz, 2022-12-31:
I tested this to work on recent builds.

Please join us on discord to get more rapid help

https://discord.gg/QWU59Tyy26

woofwoof, 2023-02-22:
I'm sorry I kind of got distracted. I got this working on a Raspberry Pi 4 absolutely no problem but it never worked on my Raspberry Pi Zero 2. I am closing the comment as that did the trick really. Thank you.

closed 2023-02-22 by woofwoof