GenPi64/gentoo-on-rpi-64bit #43: Where is the latest image?

kc2zgu, 2023-08-20:
Links to the latest images in the current readme are a little vague and inconsistent, and at least one is a 404.

https://s3.genpi64.com/images/genpi64-arm64-openrc-desktop-alpha9.img.zst seems to be from January 2022, which is a suboptimal starting point these days. Also the alpha9 branch referenced in repos.conf/genpi64 does not exist anymore. There is another closed issue about this that was closed because the user was using an old image.

https://fi.packages.genpi64.com/genpi-aarch64-desktop-latest.img.zst referenced in a wget command is 404.

Are the "nullGenPi64" images listed at https://s3.genpi64.com/images/ current? These appear to be the ones with most recent timestamps.

Openrc images on this server seem to have stopped in February while systemd is still being updated. Is openrc no longer being built?

jonesmz, 2023-08-20:
You're right, most of the readme is badly out of date.

I took a break from the project, getting back into it now.

Both openrc and systemd are intended to be supported. 

Once i get the jenkins build happy again i'll update the readme.

kc2zgu, 2023-12-31:
New builds seem to be appearing on the S3 server. Any chance you could change the build dates in the filenames to ISO6801 though so it's a little easier to see which is the newest without parsing out the modified date for each one from the XML?

jonesmz, 2023-12-31:
I haven't cut a new official build for a while as our CI pipeline is grumpy and I'm working on making it more robust.

Check on the discord channel, I have links to builds I produced on my local machine which are known to boot and can be used as a good starting point.

Your idea about the iso date format is a good one. 

Feel free to make a pull request for that, but I'll get to it eventually.

mbernardi1961, 2024-04-06:
Can you provide a viable discord link?

jonesmz, 2024-04-06:
https://discord.com/invite/cRj2TBR8zE