GenPi64/gentoo-on-rpi-64bit #44: Alpha9 Branch and Gentoo Git still being required in Lite image

Ofthemasses, 2024-01-15:
## Description

When running emerge --sync profiles are broken on the lite image. This is due to configuration files requiring the now merged and deleted branch "alpha9" from Build.Dist, and gentoo not using rsync for getting its contents.

jonesmz, 2024-01-15:
Thank you very much for the issue.

We are currently working on improvements to our continuous integration pipeline so that we can resume producing nightly builds.

The latest version of the image builder addresses this issue you have reported, but we can't link an "official" build until our continuous integration infrastructure is fixed.

If you check with us on our discord channel, i'd be happy to share an unofficial build that I produced locally.

Ofthemasses, 2024-01-16:
> Thank you very much for the issue.
>
> We are currently working on improvements to our continuous integration pipeline so that we can resume producing nightly builds.
>
> The latest version of the image [...]

Hey, ive fixed it on my current build so shouldnt need a image thanks. Just putting this issue up as it was a bit confusing to fix the problem, im assuming the build is meant to work out of the box other than timezone settings etc.

jonesmz, 2024-01-16:
> im assuming the build is meant to work

Well, it was always supposed to be an experimental build with the expectation that we'd build a "beta" and then a "release" build soonish after.

We lost some contributors over time, so not a lot is getting taken care of.

Currently I'm using the image builder to create an aarch64 VM image for our continuous integration environment, with the goal of using that as the basis for the CI. Once that's working we should be able to get nightly builds working reliably.